digmorepaka/thinkpad-firmware-patches #9: Thinkpad T14 AMD Gen1

Minssc, 2021-08-07:
| Model | T14 AMD Gen1 |
| --- | --- |
| Patchset | Gen1_AMD |
| TPM | Unknown |
| Notes | no POST, keyboard leds(FnLock, F1, F4, CapsLock) blink with 4-5 seconds interval. <br /> tried on BIOS ver.1.34 <br /> also, whitelist patch does nothing as pattern is not found at all. |

I'm willing to help test things

digmorepaka, 2021-08-07:
Did you use this https://github.com/sibradzic/UEFI-playground/blob/master/fix_vendor_hashes.py or hexedit method to disable tamper protection?

Minssc, 2021-08-07:
I tried both methods and they both doesn't post.

Minssc, 2021-08-07:
I also tried w/o tpm patch to no avail. Perhaps it's me that did something wrong. Here are the rom files

[bios images.zip](https://github.com/digmorepaka/thinkpad-firmware-patches/files/6949224/bios.images.zip)

digmorepaka, 2021-08-11:
Updated the patches from xx, this time try without the hex replacement.

Minssc, 2021-08-12:
no luck, same symptoms. FYI i had my board swapped would that matter?

digmorepaka, 2021-08-14:
Try flashing the bypass patch only to verify that our bypass method works and that it's the other patches causing problems

Minssc, 2021-08-15:
tried only patching the bypass and similar symtoms, keyboard LEDs flash in more constant 4 second interval.

digmorepaka, 2024-01-25:
Looks like gen 1 AMD, and xx90 use a completely different tamper check bypass. Until that's publicly documented we're SOL. It does exist since allservice does have their password removal module functional, and that requires it.

closed 2024-01-25 by digmorepaka