digmorepaka/thinkpad-firmware-patches #14: T440 whitelist

goldwoods2005, 2021-12-07:
I do patch my bios with the patch file provided, but after flash back to the bios chip ; It is not working.
My Bios version: 2.52 (the latest one)
I see that, The tested just on the bios 2.41.

My Dump Bios from flashrom :https://drive.google.com/file/d/1VfIPq9IYaQaMMazW0DTnYnkv889cCGB8/view?usp=sharing

Many Thanks

digmorepaka, 2021-12-07:
Rom looks fine. did you replace `4E 56 42 42 53 45 43 FB` with `4E 56 42 42 53 45 43 FF` on your patched rom?

goldwoods2005, 2021-12-07:
> Rom looks fine. did you replace `4E 56 42 42 53 45 43 FB` with `4E 56 42 42 53 45 43 FF` on your patched rom?

I did sign the patch rom with thrimbor/thinkpad-uefi-sign before flash bios.

My Patch Bios https://drive.google.com/file/d/1PqN4lJg87CbwiUOEBxVjElFQI3HoF1Zc/view?usp=sharing

digmorepaka, 2021-12-07:
That currently doesn't work on the  T440.

goldwoods2005, 2021-12-07:
I tried to downgrade bios but it can't be done.
I forgeted to mention you my devices is T440. 

So happy that you replied so quickly
Thanks

goldwoods2005, 2021-12-09:
It's me again.
So I did not patch with the code replace 4E 56 42 42 53 45 43 FB with 4E 56 42 42 53 45 43 FF, I tried the luck again, and boom. It's worked. (Try to replace with 3165 card, no problems, stable signal)

I can confirm the patch work on latest bios (2.54) Please update the text.
Many thanks.

closed 2021-12-09 by goldwoods2005

p4block, 2024-08-31:
@digmorepaka This also happened to me, I didn't get that "TPM: No" meant that I shouldn't be using sign.py 
I was pretty asleep when I was doing this, but maybe the README lacks a notice about it. Right now it tells you that sign.py is for "xx40 ThinkPads" of which this one is a part of! 

In any case, only one bios chip vs my x230 made this a breeze and I was able to get it on the second try. 
Awesome work on this, thanks a million

digmorepaka, 2025-01-20:
> the "4E 56 42 42 53 45 43 FB to 4E 56 42 42 53 45 43 FF" 

Yeah use that for T440