digmorepaka/thinkpad-firmware-patches #16: Whitelist removal not working properly on X1C6

Housies, 2022-02-01:
I'm trying to remove the whitelist on my 6th gen X1 Carbon so I can use a Sierra EM7455. I applied all the patches in xx70_xx80_patches_v6.txt, and replaced `4C 4E 56 42 42 53 45 43 FB` with `4C 4E 56 42 42 53 45 43 FF` in a hex editor, and I have the advanced menu in the BIOS, but it seems like the whitelist is (mostly) still there. Sometimes it boots fine and recognises the EM7455, and sometimes it gives me the "Unauthorized network card" error, and I don't know why. I updated the BIOS to version 1.54, so I'm wondering if it only works on older versions, or different models, or something. I double checked that I'd uncommented the right lines in the text file. Am I doing something wrong?

thinkpader-hub, 2022-02-06:
Hi,

I have same issue with bios 1.55 on X1C6, still can not find the solution.

if someone knows, please help me.

Thank you for your help.

digmorepaka, 2022-04-19:
I'll take a look, could you post your dump?

Housies, 2022-05-06:
Yes, it's here: https://www.mediafire.com/file/o1fu3bq9pi8ztz2/patchedbiosreplace.bin/file

OhYou-0, 2022-06-08:
Having similar issue with t460s, em7411 modem
em7565 modem works fine after patches.
foxconn T77H468 also works fine after patches 
tested the 7411 in two different patched t460s laptops, same results. 

very confusing to me why this would be only a partial whitelist removal it seems. 
The 7411 is more ideal for my carrier so it would be nice to get it working.
I'm not sure if this would be the same issue as OP though, Maybe try different modems if you have any? I have a em7455 but as the laptop shipped with them, it obviously works without issue

Some0nee, 2024-11-18:
Can you also post a clean dump of your image, without all modifications?

ulisesvina, 2025-01-12:
Same issue can be reproduced with a ThinkPad X380 Yoga and X1 Yoga gen. 2 and the Fibicom L830-EB card.

Some0nee, 2025-01-20:
@ulisesvina

Please post a copy of your dump and original dump.

ulisesvina, 2025-01-20:
Sure. My dump is here: [https://drive.google.com/file/d/1nH7ZJYhkcngWGaza7r0UJk9xAgcqVq-S/view?usp=sharing](url)

will127534, 2026-04-05:
I have this one working for me: https://github.com/will127534/thinkpad-firmware-patches/blob/master/x1c6_patches_v1.txt