digmorepaka/thinkpad-firmware-patches #18: WWAN whitelist removal not working on T480s

TitoBournet, 2022-04-03:
Model | T480s
-- | --
Patchset | xx70_xx80_patches_v6.txt (only advance menu and wwan whitelist selected)
TPM | MFG state
Notes | I have the advanced menu in the BIOS, although the WWAN whitelist is still there.

Hello everyone!  
I'm trying to remove the WWAN whitelist on my T480s.  

I've applied the patches "enable advance menu" and "remove wwan whitelist" in **xx70_xx80_patches_v6.txt**  
To disable tamper protection I've replaced `4C 4E 56 42 42 53 45 43` **`FB`** with `4C 4E 56 42 42 53 45 43` **`FF`** in a hex editor, and after flashing it back, I have the advanced menu in the BIOS, but the WWAN whitelist is still there. If I leave the WWAN card inserted, I get error 1802:  

`1802: Unauthorized network card is plugged in - Power off and remove the network card`  
`System is halted`  

TPM was previously disabled, but after the patch, it was in "**MFG**" state.  

Should I use a different script on my T480s?  
Could someone help me to remove the WWAN whitelist?  

Thank you very much!

digmorepaka, 2022-04-19:
I'll take a look, could you post your dump?

TitoBournet, 2022-04-21:
Hi Daniel,

Here is a dump of my original [BIOS ROM:](https://www.mediafire.com/file/r8x1cjxauj3o4n4/T480s_bios_orig.img/file)

Thanks in advance,
Pablo

digmorepaka, 2022-05-05:
I see absolutely zero why the regular patches wouldn't work, nonetheless, here's a manually modified version only containing the wwan whitelist bypass. http://0x0.st/oARB.bin md5:`e3737ff1846f47aebc346df0d4f0cfde  T480s-patched-UEFI.bin`

digmorepaka, 2022-05-05:
woops

TitoBournet, 2022-05-06:
Hi Daniel, thank you very much!

I'll try it as soon as I get a chance (sometime over the weekend I estimate) and let you know.

Best regards,
Tito

TitoBournet, 2022-05-08:
Hi,
I flashed back the BIOS MOD that Daniel (@digmorepaka) did and I still get the same behavior as doing it through the scripts, i.e. if I leave the WWLAN card plugged, I get error 1802:

`1802: Unauthorized network card is plugged in - Power off and remove the network card`
`System is halted`

Even the behavior of the TPM was the same (from disabled to MFG state).

What other test could I perform?

Thanks a lot,
Tito

digmorepaka, 2022-05-11:
The module looks exactly the same as the one in the T480 UEFI, for  some reason the pattern isn't automatically  found - ok  found it manually. But it does not work, might need a second module patched too. If I get around to it, I will post it here. Currently working on a couple other projects so sorry if I take a while to do it.

TitoBournet, 2022-05-12:
OK Daniel, thank you very much for taking the time.
There's no rush, so I'll wait until you get a chance to see it.
Best regards,
Tito

k1gen, 2023-05-22:
hey @digmorepaka, I have the same issue on my T480s after applying Advanced menu and WWAN whitelist removal patches: Advanced menu appears and is usable, but WWAN whitelist is still there. any chance you could fix this with now three of us encountering this exact problem? I can send my original and current BIOS dumps if you need them

k1gen, 2023-07-08:
@digmorepaka? you there?

maz3max, 2023-07-25:
Just noticed that this specific error message is not from the LenovoWmaPolicy module, but from the BsdMilestone module. Still have no clue how to
 properly patch this.

Attaching the extracted efi file in case anyone would like to take a stab at it:
[Section_PE32_image_BdsMilestone_BdsMilestone_body.efi.zip](https://github.com/digmorepaka/thinkpad-firmware-patches/files/12155943/Section_PE32_image_BdsMilestone_BdsMilestone_body.efi.zip)

maz3max, 2023-07-25:
Maybe the line `6A628EFE-3682-4FDC-A31E-C635BDF18CC8 10 P:803D1CDE0000000F841102:803D1CDE000000E9120200` could fix it. No guarantee though, I need to wait for hardware to test it. Assume that this damages your system.

maz3max, 2023-07-26:
Just noticed that I don't need to mod the bios of my X1C5 to install an AX210 card, so I won't pursue this further.

k1gen, 2023-07-26:
I still want @digmorepaka or someone else to fix this, because for now any adequately-priced WWAN card won't work in my T480s

dukzcry2, 2024-04-09:
I can't make universal patch, but here is what works for me for 1.57 bios
https://gitlab.com/repos-holder/thinkpad-firmware-patches/-/blob/master/xx70_xx80_patches_v7.txt?ref_type=heads
Its not hard to adjust it for specific bios version with ida pro. https://gitlab.com/repos-holder/thinkpad-firmware-patches/-/blob/master/notes.txt?ref_type=heads may help