digmorepaka/thinkpad-firmware-patches #20: T495 supported instead of "probably"

OhYou-0, 2022-05-18:
| Model | T495|
| --- | --- |
| Patchset | Default for generation |
| TPM | Yes, fix_vendor_hashes |
| Notes | Desoldering rom required, or possibly isolate programmer vcc and flash running system, similar to x60 |

digmorepaka, 2022-05-18:
Awesome, also is the flash 1.8V or 3.3V?

OhYou-0, 2022-05-18:
1.8v

closed 2022-05-18 by digmorepaka

AkechiShiro, 2023-02-12:
Hi @OhYou-0, I've got a few questions if that's okay, the most important one is the last one the 6. question, if you have any idea about that, it would be extremely helpful to me.

1. Could you explain in more details or link to any resources on how to do that with more details and maybe pictures (if you happen to know any resources), how would it be possible to isolate the programmer VCC and the flash running system without desoldering and without breaking/cutting any traces on the motherboard ? 
2. Could you also precise the flash chip exact name (probably a Winbond something) ? 
3. Have you tried lowering the voltage lower than 1.8V (may be unstable but can still work without getting other components on I guess?)
4. Flashing allowed you what exactly (AMD menu unlock to get what kind of precise features that were locked ?)
5. Which patches did you flash, [all the patches included here?](https://github.com/digmorepaka/thinkpad-firmware-patches/blob/master/xx85_xx95_patches_v1.txt)

I've been trying to get an SSD working on the WWAN port (it works in the WWAN port of T450 without any flashing because there is no whitelisting on that model), but on the T495 it is not being detected by the BIOS, right now, I believe the whitelisting is the culprit forbidding this, but I may be wrong.

6. Can you confirm (if you have ever tested) that on your (bios unlocked) T495, that an SSD 2242 PCIe NVMe in the WWAN port works with the whitelist removed ?

OhYou-0, 2023-02-13:
The method for the x60 flashing is to connect up the programmer, but leave off the vcc pin, then power up the board, which powers the rom and you can then flash it. 
I think this may work with the T495 but did not test it as when I disconnected my programming clip from the rom, it actually broke the soic8 off of the board cleanly. I guess my solder joints were very weak from factory. 
I did also find out much later that the chip is 1.8v and was having difficulty flashing it on board with 3.3v, as it would cause the rpi to shut off, so I think the board is attempting to power up components from the power input.

flash chip is a 25q128fwsq
flashing allows the amd menu, but some of the nice features do not work unfortunately. 
I was able to set the target TDP up to 54w which gives a decent boost in performance. There is no control over short term or long term boost though, which is where you can make some serious gains. You can do all of this in the os with software but its much better done from the bios. 
You can also adjust the above 4gb mimo limit which is nice.
There is also some other stuff like switching the m.2 from x2 to x4 (i have no ides why it is x2 default) but I never tested it, and also memory timings can be adjusted but it does not seem to make a difference. 

I flashed the one as linked. 
as for the ssd, I do not have one to test at the moment, although I am interested myself so I bought one. I'll check when it comes tomorrow.

OhYou-0, 2023-02-14:
@AkechiShiro Tested, working. 
Drive is KBG30ZMT256G fru: 00UP651
Drive runs as PCI-e 3.0 x1 speed unfortunately.

AkechiShiro, 2023-02-18:
Thanks @OhYou-0 for all the info, I actually am going to buy this https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all and plug the RPI 3.3V onto one side, and the 1.8V onto the other side of it, but basically it should convert the SPI 3.3V lines to 1.8V lines and thus the flashing should go alright.

What kind of RPI (Raspberry Pi) were you using, model 3B+ ? model 4? or 1 perhaps ?