digmorepaka/thinkpad-firmware-patches #21: T470S Firmware patch

hamcuks, 2022-06-12:
Hi, can you make patch for t470s? I've apply the patch (xx70_xx80_patches_v7) to my t470s BIOS, but no luck

My laptop beeping 5 times, but not boot to BIOS or OS. Just black screen

Thanks in advance!

digmorepaka, 2022-06-12:
> replace 4C 4E 56 42 42 53 45 43 FB with 4C 4E 56 42 42 53 45 43 FF

Have you done  this  step?

hamcuks, 2022-06-12:
> > replace 4C 4E 56 42 42 53 45 43 FB with 4C 4E 56 42 42 53 45 43 FF
> 
> Have you done this step?

Oh sorry, I forget to replace that. Let me try, I'll report later...

hamcuks, 2022-06-12:
> 

Sorry, I don't understand how to replace the hex number. Using UEFIReplace or what?

Thank you

digmorepaka, 2022-06-12:
.....a hex editor

hamcuks, 2022-06-25:
Hi, I'm sorry for late reply. Now, I success unlock the advanced tab. I was forgot to replace replace 4C 4E 56 42 42 53 45 43 FB with 4C 4E 56 42 42 53 45 43 FF

[image]

closed 2022-06-25 by hamcuks