digmorepaka/thinkpad-firmware-patches #22: Thinkpad X1 yoga gen 3

Minssc, 2022-08-17:
| Model | Thinkpad X1 yoga gen 3 |
| --- | --- |
| Patchset | xx80 |
| TPM | Yes |

RimbyCarkey, 2022-08-26:
I am having trouble with this `rimby@rimby-ThinkPad-X1-Yoga-3rd:~/Downloads/UEFIPatch_0.25.0_linux_x86_64$ ./UEFIPatch /home/rimby/LenovoCFG/N25ET61W/firmware.bin /home/rimby/Downloads/LenovoPatch.txt -o  /home/rimby/LenovoCFG/N25ET61W/firmwareAdvanced.bin 
parseVolume: unknown file system 00504624-8A59-4EEB-BD0F-6B36E96128E0
parseVolume: unknown file system FFF12B8D-7696-4C8B-A985-2747075B4F50
parseBios: one of volumes inside overlaps the end of data
parseFile: non-empty pad-file contents will be destroyed after volume modifications
Function called with invalid parameter
rimby@rimby-ThinkPad-X1-Yoga-3rd:~/Downloads/UEFIPatch_0.25.0_linux_x86_64$`

Minssc, 2022-08-26:
I too get those warnings but the patch continues fine after the parseFile warning

`parseVolume: unknown file system 00504624-8A59-4EEB-BD0F-6B36E96128E0
parseVolume: unknown file system FFF12B8D-7696-4C8B-A985-2747075B4F50
parseBios: one of volumes inside overlaps the end of data
parseFile: non-empty pad-file contents will be destroyed after volume modifications
patch: replaced 16 bytes at offset 118D0h 04320B483CC2E14ABB16A73FADDA475F -> 778B1D826D24964E8E103467D56AB1BA
patch: replaced 28 bytes at offset 6B8BEh 09072C1300000209072D1300000309072E1300000409072F13000005 -> 09072C1300001909072D1300001A09072E1300001B09072F1300001C
patch: replaced 58 bytes at offset 53C04h 00143100300030004D0068007A000000143100350030004D0068007A000000143200300030004D0068007A000000143200350030004D0068007A -> 00143100320035003000200020000000143100330030003000200020000000143100330035003000200020000000143100340030003000200020
patch: replaced 26 bytes at offset 69E91h 0A821286A10B04001206A60B010016021206A60B020016022902 -> 0A821286A10B04001206A60B000016021206A60B000016022902
patch: replaced 11 bytes at offset 545h 0BC8394B040F8437020000 -> 0BC8394B04E93802000000
patch: replaced 5 bytes at offset 535h C8390B7516 -> C8390B7500
Image patched`

Not sure where that Function called with invalid param came from

RimbyCarkey, 2022-08-26:
I am only trying to enable the Advanced BIOS menu. I updated as i was on 0.25.0 and i got the same result. I want to say it might be the firmware file but we both have the same firmware file. Just to make sure, the first file (for the BIOS not the controller) is N25ET61W.

RimbyCarkey, 2022-08-26:
> I too get those warnings but the patch continues fine after the parseFile warning
> 
> `parseVolume: unknown file system 00504624-8A59-4EEB-BD0F-6B36E96128E0 parseVolume: unknown file system FFF12B8D-7696-4C8B-A985-2747075B4F50 parseBios: one of volumes inside overlaps the end of data parseFile: non-empty pad-file contents will be destroyed after volume modifications patch: replaced 16 bytes at offset 118D0h 04320B483CC2E14ABB16A73FADDA475F -> 778B1D826D24964E8E103467D56AB1BA patch: replaced 28 bytes at offset 6B8BEh 09072C1300000209072D1300000309072E1300000409072F13000005 -> 09072C1300001909072D1300001A09072E1300001B09072F1300001C patch: replaced 58 bytes at offset 53C04h 00143100300030004D0068007A000000143100350030004D0068007A000000143200300030004D0068007A000000143200350030004D0068007A -> 00143100320035003000200020000000143100330030003000200020000000143100330035003000200020000000143100340030003000200020 patch: replaced 26 bytes at offset 69E91h 0A821286A10B04001206A60B010016021206A60B020016022902 -> 0A821286A10B04001206A60B000016021206A60B000016022902 patch: replaced 11 bytes at offset 545h 0BC8394B040F8437020000 -> 0BC8394B04E93802000000 patch: replaced 5 bytes at offset 535h C8390B7516 -> C8390B7500 Image patched`
> 
> Not sure where that Function called with invalid param came from maybe try with newer uefipatch? I think I used 0.28.0

I am only trying to enable the Advanced BIOS menu. I updated as i was on 0.25.0 and i got the same result. I want to say it might be the firmware file but we both have the same firmware file. Just to make sure, the first file (for the BIOS not the controller) is N25ET61W

Minssc, 2022-08-26:
I believe the error is not patch related. Try cleaning up the parameter a bit or I could try patching your dump for you

RimbyCarkey, 2022-08-26:
> I believe the error is not patch related. Try cleaning up the parameter a bit or I could try patching your dump for you

I don't see a way to clean it up as it is so simple but i have uploaded the files to [google drive](https://drive.google.com/drive/folders/1wPhI6InMdDRB3Gn5nb9izO887v3F4oZ0?usp=sharing). Most of the files are self explanatory however i also put in a .sct file which contains where the cfg lock is. My goal out of this is to disable it using the advanced menu.

Minssc, 2022-08-26:
Oh there is a cfg lock option in the advanced bios. I remember seeing it one moment pls

RimbyCarkey, 2022-08-26:
> ah you've un-commented line #27 of your patchfile which was causing the issue. you should be able to patch if you comment line #27. Here's the patched bin anyways oh and make sure to hex edit the tpm thing [firmware.bin.zip](https://github.com/digmorepaka/thinkpad-firmware-patches/files/9434212/firmware.bin.zip)

May i ask what you used to patch the bios?

Minssc, 2022-08-26:
ueftpatch 0.28.0 with your patchfile with line#27 commented out

RimbyCarkey, 2022-08-26:
> ueftpatch 0.28.0 with your patchfile with line#27 commented out

Pardon my wording, I meant to update the BIOS to the custom .bin.

Minssc, 2022-08-26:
oh. lol. uhh i used ch341a spi flasher

RimbyCarkey, 2022-08-26:
> oh. lol. uhh i used ch341a spi flasher

Wow that is a lot more involved than i thought. I was under the belief that we could put it on a usb drive and use an updater. Maybe fwupd is the answer here.

Minssc, 2022-08-26:
to my short knowledge thinkpad bioses are somehow temper protected and needs to be patched physically using spi flasher

digmorepaka, 2022-08-26:
The updater part of the UEFI rejects any deployed package that is invalid. Either modified or corrupt from a bad download.

RimbyCarkey, 2022-09-04:
> to my short knowledge thinkpad bioses are somehow temper protected and needs to be patched physically using spi flasher

Completly unrelated but have you made a hackintosh with your X1 yoga?

Minssc, 2022-09-04:
no

githubbapoopa, 2022-12-26:
> ah you've un-commented line #27 of your patchfile which was causing the issue. you should be able to patch if you comment line #27. Here's the patched bin anyways oh and make sure to hex edit the [...]

Any hint, info or heads-up on "hex editing" the TPM on the X1Y3 would be much appreciated.

I assume Fingerprint still works then with the patched Bios?

Were you able/did you try to use a different WWAN than the Fibocom (Quectel EM120R-GL or Qualcomm Snapdragon X55 5G)?

Minssc, 2022-12-27:
> Did you simply replace 4C 4E 56 42 42 53 45 43 FB with 4C 4E 56 42 42 53 45 43 FF to get the TPM working?

yes
> I assume Fingerprint still works then with the patched Bios?

yes
> Were you able/did you try to use a different WWAN than the Fibocom (Quectel EM120R-GL or Qualcomm Snapdragon X55 5G)?

I did not have any other module than L850 at the time but it was not whitelisted part, so I'd assume whitelist removal worked fine.
> Did you try the patch on the latest 1.48 Bios?

I belive it was 1.46 or 1.47. I do not have the device on hand anymore.

githubbapoopa, 2022-12-27:
> yes
>
>
> yes
>
>
> I did not have any other module than L850 at the time but it was not whitelisted part, so I'd assume whitelist removal worked fine.
>
>
> I belive it was 1.46 or 1.47. I do not have the [...]

The newest could have been 1.47 at the time of your post. I assume the patch is working on all bios versions then.

Will report back on WWAN whitelist as soon as I got my hands on a Quectel EM120R-GL.

Thanks so much!

githubbapoopa, 2022-12-27:
> > to my short knowledge thinkpad bioses are somehow temper protected and needs to be patched physically using spi flasher
> 
> Completly unrelated but have you made a hackintosh with your X1 yoga?

Neither did I but as starters you could have a look here:

https://github.com/Jamesxxx1997/thinkpad-x1-yoga-2018-hackintosh

https://github.com/tylernguyen/x1c6-hackintosh