digmorepaka/thinkpad-firmware-patches #26: W541 is not seeing m2 ssd after bios patch

gassity, 2022-12-07:
Everything is working fine, except  for m.2 ssd, which worked good before patching.
No matter how I try. With every bios settings I've tried.
It's not being detected both in BIOS and Windows.

SSD is working fine itself. Checked in another laptop.

What could be the problem?

digmorepaka, 2022-12-07:
Make sure the sata port that it is attached to is enabled of course start by loading defaults.

gassity, 2022-12-08:
> Make sure the sata port that it is attached to is enabled of course start by loading defaults.

yep, sure. Already done, but no luck. Maybe there's some specific setting in "Advance menu"?

digmorepaka, 2022-12-08:
> Maybe there's some specific setting in "Advance menu"
Yes, check if the sata port is enabled.

gassity, 2022-12-08:
Sure) It sees my other 2.5 SSD and DVD-Rom. I've tried to even unplug them for any interference, but still no luck. Led on the m.2 ssd lights up and that's it.

digmorepaka, 2022-12-08:
/shrug

closed 2022-12-08 by digmorepaka

dnsound, 2023-04-18:
Same issue for me, did the bios patching to my W541 for the advanced menu and my m.2 ssd is no more detected.
I also tried booting with coreboot but the issue persist.