digmorepaka/thinkpad-firmware-patches #27: Bricked X240: understanding what went wrong

Oposum01, 2022-12-07:
Hey @digmorepaka , I bricked a X240 :D
Just wanted to make sure, I do understand what went wrong to prevent this from happen to a newly bought X240 motherboard.

The device came with BIOS v2.25, which has been upgraded by myself to v2.49 using an external CD drive and the official BIOS utility. Embedded controller got updated to v1.18 sucessfully, too.
After dumping that image, I flashed a modified one which included only the patches from https://github.com/digmorepaka/thinkpad-firmware-patches/blob/master/xx40_xx50_patches_v7.txt - result: devices booted successful (I was able to see the Advanced menu in the BIOS) but with (not absolutley sure anymore) 5 beep tones. I thought that's the tamper protection complaining (but it seems to be the chassis intrusion only?).

From here everything went wrong: So is started to sign the image with "Thrimbor's uefi-sign" and flashed that image. Display wasn't working anymore, though the keyboard backlight turns on, even the fan for a short time, but I got no more beep tones. Flashed a BIOS image which included only the replacement of "4C 4E 56 42 42 53 45 43 FB" with "4C 4E 56 42 42 53 45 43 FF" (besides the patches, but no uefi-sign) with the same outcome. I read back the image several times to verify it's the one written to it (even though flashrom does that for me, just wanted to double check) - turned out it has the same md5sum. Flashed the patched one, which worked initially, same outcome. Flashed the initial v2.49 without any modifications - same outcome, still bricked. Flashed the v2.25 which I dumped too in the beginning, same outcome: still bricked.

My question: can you think of any recovery method to this issue? Or do you have an idea what could have gone wrong?
"uefi-sign" is not needed for X240, bc TPM is not working? I don't care about TPM, aiming to run Linux.
Do I really have to replace the "4C 4E 56 42 42 53 45 43 FB" with "4C 4E 56 42 42 53 45 43 FF" for the X240 (Intel i5-4210U)?

Thanks in advance :)

digmorepaka, 2022-12-07:
Either damaged the board while flashing or IME screwed itself up. Disconnect all batteries for about 10 minutes. Afterward connect charger and hold power button for 30 seconds, this clears the EC, and should bring you into a full cold boot state.

digmorepaka, 2022-12-08:
thinkpad-uefi-sign finds the tamper protection checsum and re-generates it, works on xx30/xx40(M, not U) usually. hex string turns the tamper protection off which also disables the TPM until a rom with the correct checksums is flashed. On models  up to xx40 or xx50 I believe you get 5 angry beeps and no TPM function when  the  checksums are wrong.

Oposum01, 2022-12-13:
Thanks for the explanation
So I am going to summarize the right procedure for the X240, which should be:
1. Upgrade the X240 to official v2.49 BIOS
2. Dump the BIOS via SPI Flasher
3. Patch the dumped file with xx40_xx50_patches_v7.txt:
  UEFIPatch <bios-dump.file> xx40_xx50_patches_v7.txt
4. Open the patched BIOS file with a HexEditor and replace "4C 4E 56 42 42 53 45 43 FB" with "4C 4E 56 42 42 53 45 43 FF" to turn off the tamper protection (which disables TPM too) to get rid of the "5 angry beeps"
5. Write the modified BIOS file (containing the patches and the hex replacement) to the X240 via SPI Flasher

Actually I was _not_ able to recover my X240 motherboard. Holding power button and/or reset button (tiny hole on the base cover) for 30 seconds in the hope to reset the embedded controller didn't do the trick. I purchased a second motherboard, where I am stuck right now too, because it's not working (didn't even modify the BIOS). But that should be not your problem ;). I'll comment when I got the board or another replacment working (incl. the BIOS patching procedure).

Oposum01, 2022-12-22:
Finally the second replacement board came in and worked out fine. I am able to confirm the procedure from my previous post works. I do not get any angry beeps. I recommend booting into UEFI after flashing and loading the default settings + doing some full boot cycles into a OS and so on, in case the embedded controller needs some time to re-adjust.

Remark 1: I've removed the CMOS battery during flashing. The notebook was missing a correct date and time when booting then (could be workaround pressing F1): I was not able to find the Date/Time settings in BIOS anymore. Luckily Ubuntu will set the hwclock in the OS for me.

Remark2: I get a "2101: Detection error on SSD1 (M.2)" error during booting, when a Fibocom 850 LTE card is installed to the (bigger) m.2 socket (workaround "Press ESC to continue"). Don't know why it's complaining, I was able to see the card in Ubuntu though.

All in all my issue is resolved. My X240 runs with a MediaTek MT7921K Wi-Fi card now.

closed 2022-12-22 by Oposum01