digmorepaka/thinkpad-firmware-patches #34: WWAN whitelist removal does not work on X270

vbooka1, 2023-04-28:
Hello, my issue is very same as #18 - I have dumped the original firmware, applied the patch, replaced `4C 4E 56 42 42 53 45 43 FB` with `4C 4E 56 42 42 53 45 43 FF`, flashed the patched firmware, got the Advanced Menu with overclocking and all other stuff, but the WWAN whitelist is still in force - the laptop does not boot and shows error "1802: unauthorized network card" with a 3rd party card. If I put the original WWAN card the laptop boots normally.

Also there is another small issue - the laptop complains about incorrect time/date and wants me to adjust it but I could not find where to change the time/date in the patched BIOS. It is not on the 2nd tab as it was in the original BIOS (as far as I remember) and I wasn't able to find it anywhere else, could you advise how do I change the time/date?

The full model name is: 20HN0012UK SDK0J40697

vbooka1, 2023-04-30:
> Also there is another small issue - the laptop complains about incorrect time/date and wants me to adjust it but I could not find where to change the time/date in the patched BIOS. It is not on the [...]

okay searching for possible solutions I've found this repo for older Thinkpads: https://github.com/n4ru/1vyrain and it addresses this issue in FAQ:

> Missing Date & Time in BIOS
> 
> This is a side effect of the Advanced Menu needing to go somewhere. You’re not blind. The Date & Time tab is gone. If you really need it, flash a fresh BIOS. You probably don’t though. Setting this in the OS propagates it to the BIOS, in both Windows and (all?) flavors of Linux.

vbooka1, 2023-05-28:
still interested, will pay for a solution.

vbooka1, 2023-06-04:
I've tried to replace the original modem's vendor ID and product ID with a new modem's IDs in a hex editor, but it did not help, as well as patching the updated dump with this repo patches. 

I've replaced about 10 occurrences of original modem's IDs with the new IDs:
[image]

but laptop still reports error 1802. Then I've put back the original modem and booted the laptop with patched BIOS, and it booted normally, having no mentions of the original modem's vendor ID and product ID. 

This means that the real whitelisted IDs are somehow hidden or encrypted.

fipti, 2023-07-11:
Hi! I am also interested. Have you found a working patch?

vbooka1, 2023-07-21:
No I did not, but I've got two more ideas: extract the LenovoWmaPolicyDxe using the UEFITool, and:
1) either replace the vendor/product IDs in the extracted part (I did that already but did not test the patched BIOS image yet)
2) or disassemble the part and modify the jumps in the assembly (I've never done that but Google suggests that it's not very hard)

Also you could test your luck here https://www.bios-mods.com/forum/Thread-REQUEST-Lenovo-X270-R0IETxxW-Whitelist-Removal with the fucking snob Dudu2002, but he is very picky about who to help, for example he stopped replying to me after I told him my laptop serial numbers.