digmorepaka/thinkpad-firmware-patches #42: Unable to patch t480

feathervoid, 2024-01-16:
Tried to patch t480 twice, and all I got was error beeps. Used all relevant patches, unable to find hex string `4C 4E 56 42 42 53 45 43 FB` in patched BIOS. Thinkpad is on latest firmware update

feathervoid, 2024-01-17:
patched it again, finally got access to the advanced menu! Thank you to the folks that put in all the effort to get this data out to the public

closed 2024-01-17 by feathervoid

mjok0, 2024-02-13:
Have the same issue. What did you exactly do?

mjok0, 2024-02-13:
010288 error tone

feathervoid, 2024-02-15:
Hey @mjok0 , sorry I haven't gotten back to you sooner. I used [this guide](https://www.reddit.com/r/thinkpad/comments/16v26yl/guide_thinkpad_bios_reading_patching_and_flashing/) from reddit. Hope it helps and good luck flashing your BIOS!