digmorepaka/thinkpad-firmware-patches #44: WWAN whitelist removal not working on T560

starslab, 2024-02-11:
I'm trying to add an NVMe SSD to the WWAN slot on my T560, and getting the "1802: Unauthorized network card is plugged in" error. I've tried patches from "xx60_patches_v1.txt", "xx40_xx50_patches_v7.txt", and "xx70_xx80_patches_v7.txt". All of them either do not apply in UEFIPatcher or fail to remove the whitelist error when flashed to the machine.

I'm going to hazard a guess this is related to the several other outstanding requests here for help with WWAN issues.

Anyone have any ideas? Lenovo is doing a really good job of ensuring my next notebook purchase is from Framework.

starslab, 2024-02-13:
As an update, an experienced BIOS modder over on bios-mods.com managed to remove the whitelist, but it appears there's no PCI-express connectivity to the WWAN socket - this makes sense as both of the approved Lenovo WWAN options are USB-over-M.2 devices.

I'm left wondering how the stock BIOS is detecting the NVMe device and refusing to boot the computer though.

digmorepaka, 2024-08-01:
NVMe can't work on that slot, only USB, SSIC(essentially usb 3.x), SIM card, and i2c is hooked up. Guessing it managed to use the i2c to communicate as to what it is. The PCIe and USB device whitelist are two separate checks so it could be jumping to the PCIe check.

closed 2024-08-01 by digmorepaka