digmorepaka/thinkpad-firmware-patches #47: P53 NVME whitelist?

dagbdagb, 2024-03-19:
I am trying to figure out if the NVME slots of the Lenovo P53 have some sort of whitelist applied. It appears nobody has been able to run an eGPU via the NVME slots on a P53 yet. Is it technically possible to create a whitelist on a PCIe/M.2 slot which says "only allow storage devices to connect"?

And if this is indeed the case, is it possible to modify and flash the P53 BIOS to remove this limitation?

digmorepaka, 2024-03-20:
I doubt it, I would *guess* that it's more of a UEFI limitation if there is one. Lenovo whitelists are pretty much never silent. I would suggest you start with a 'simpler' device such as a LAN/WLAN card.

dagbdagb, 2024-03-21:
Thank you for responding. May I ask about the potential nature of said UEFI limitation?  

The exact error quoted is: 

`2101 : Detection error on Storage Device0 (M.2)`
 
[link](https://egpu.io/forums/builds/2019-15-lenovo-thinkpad-p53-quadro-t1000-9th6ch-rtx-3070-32gbps-usb4-jeyi-leidian-adt-link-r43sg-win11-22h2/)

I also se reports of 2104, but then for P52. Not sure if these devices share error codes.

dagbdagb, 2024-03-22:
Thank you for responding. May I ask about the potential nature of said UEFI limitation?  And the potential for working around it? I do not actually have a P53 yet, but I see an opportunity for making good use of one. Hence me doing some research first.

The exact error quoted is: 

`2101 : Detection error on Storage Device0 (M.2)`

Does this pinpoint the source of the problem?

[link](https://egpu.io/forums/builds/2019-15-lenovo-thinkpad-p53-quadro-t1000-9th6ch-rtx-3070-32gbps-usb4-jeyi-leidian-adt-link-r43sg-win11-22h2/)

I also se reports of error 2104, but then for P52. Not sure if these devices share error codes.

dagbdagb, 2024-04-16:
I have verified that there isn't any whitelist. eGPU via M.2 is possible.

closed 2024-04-16 by dagbdagb