digmorepaka/thinkpad-firmware-patches #49: Lenovo Thinkpad T480 WWAN issue and Intel Boot Guard

skullface101, 2024-05-21:
I successfully patched the Bios However...
1-I cannot remove Intel ME completely due to Intel Boot Guard as it beeps 5 times in a loop
2-The WWAN whitelist removal inst working or is partially working, when i plugged in the Dell EM7455 its not recognized and i cant flash it with generic firmware.

Please help me with the WWAN patch that works properly and help me disable Intel Boot Guard.
Note: TPM is working in MFG mode and i need it for windows 11...etc

digmorepaka, 2024-05-22:
>  1-I cannot remove Intel ME completely due to Intel Boot Guard as it beeps 5 times in a loop
Bootguard has nothing to do with iME modification. You cannot completely remove iME on a 300 series chipset. 

>  2-The WWAN whitelist removal inst working or is partially working, when i plugged in the Dell EM7455 its not recognized and i cant flash it with generic firmware.

EM7455 is a USB card. T480 does not support this in hardware. 

Do some research before blindly buying parts and messing with firmwared

closed 2024-05-22 by digmorepaka

skullface101, 2024-05-22:
No
Em7455 is the same card found in T460 and T470
and i have Unlocked lenovo branded EM7455 that works even without patching the BIOS

skullface101, 2024-05-22:
The whitelist is not working

skullface101, 2024-05-22:
Reopen it

digmorepaka, 2024-05-22:
>  Em7455 is the same card found in T460 and T470

Yes, these models support USB WWAN, T480 does not.
T480 needs a PCIe card such as Intel XMM 7262.

skullface101, 2024-05-22:
EM7455 is usb over PCIe and its working in my T480
I have another one which is not unlocked
I need to unlock it

skullface101, 2024-05-22:
all my cards are M.2 WWAN form factor
and all same model with different Brand/FRU
I have lenovo T460 FRU and Lenovo T470 FRU and Dell <some model>FRU
and btw i once had an unlocked DELL EM7455 working in my T480 but i swapped it wth a lenovo one with my friend who has a dell laptop

I figured out that was a mistake because dell version supports 2 more 4G bands

skullface101, 2024-05-22:
Oh btw i have the same issue with my T470, the whitelist makes me unable to work with the dell card

digmorepaka, 2024-05-22:
> all my cards are M.2 WWAN form factor

M.2 is a physical form factor not an interface, all the documentation i found online points to them being usb only and using the m.2 slot's USB ports directly. 

If you need configure the card differently than it currently is, you need to figure that out yourself as this seems more of a card issue than a whitelist removal issue.

skullface101, 2024-05-22:
the cards works in dell laptops and usb WWAN adapters