digmorepaka/thinkpad-firmware-patches #52: IGPU oc patches doesnt work in E580

md500, 2024-09-05:
| Model  | E580 |
| ------------- | ------------- |
| PatchSet  | xx70_xx80_patches_v7.txt|
| OC igpu  | It shows options but i doesnt work. Selected 1400 but it work as 250... |
| Notes | I think it doesn't work because the first part of oc patch failed but 2nd patch worked |
| Bios - IC version | 1.38 - 1.38 |

Hi guys i tried to apply the igpu patch with the dumped bios using my ch341a, tried to patch, patches work partially but it doesnt work correctly. i think it works as 250MHz because my fps is too low...

Default Config:

[image]
[image]

1400Mhz Config:
[image]
[image]

Could someone help me to apply igpu OC?

Here is default bios:

[E580 1.31 Bios](https://www.mediafire.com/file/nvjkvkko1y4isbt/%25240AR0P00.FL1/file)

Thank you very much!

digmorepaka, 2024-09-05:
Surprising that it works on E580 at all. Try changing frequency via XTU instead of the UEFI setup.

md500, 2024-09-07:
@digmorepaka  I dont think the limitation is due to CPU, because i tried the patch with T580 bios variant (just to test if the problem is related with CPU) because T580 comes with i7-8550U.

But my E580 comes with i7-8550U too!
[image]

Output in UefiTool with T580 bios:

```
Hex pattern "09072C1300000209072D1300000309072E1300000409072F13000005" found as "09072C1300000209072D1300000309072E1300000409072F13000005" in Setup/PE32 image section at header-offset 6B8F2h
Hex pattern "00143100300030004D0068007A000000143100350030004D0068007A000000143200300030004D0068007A000000143200350030004D0068007A" found as "00143100300030004D0068007A000000143100350030004D0068007A000000143200300030004D0068007A000000143200350030004D0068007A" in Setup/PE32 image section at header-offset 53CA8h
```

Output from UefiTool with E580 bios:

```
Hex pattern "09072C1300000209072D1300000309072E1300000409072F13000005" could not be found
Hex pattern "00143100300030004D0068007A000000143100350030004D0068007A000000143200300030004D0068007A000000143200350030004D0068007A" found as "00143100300030004D0068007A000000143100350030004D0068007A000000143200300030004D0068007A000000143200350030004D0068007A" in Setup/PE32 image section at header-offset 526AAh
```

Logs from UefiPatch:

[output-T580.log](https://github.com/user-attachments/files/16919943/output-T580.log)
[output-E580.log](https://github.com/user-attachments/files/16919944/output-E580.log)

Intel XTU doen't install

[image]

But i forced the XTU install with https://github.com/gimdh/Intel-XTU-Install-Helper?tab=readme-ov-file to install the software.

I'm going to reboot to check if it works from XTU, altougth i dont think that it'll work because patcher doesnt find `09072C1300000209072D1300000309072E1300000409072F13000005` while with the other model (T580) it does find that hexadecimal pattern. 

[image]

md500, 2024-09-08:
Installed v6.5 version, because XTU options in >=v7.6 are too limited. and if i install the latest version, doesnt find the drivers.

I had to follow this [guide ](https://web.archive.org/web/20190109092500/https://kallelilja.com/2018/12/workaround-xtu-attempted-to-install-on-an-unsupported-platform/) to install Xtu with:

`msiexec /i XTUInstaller.msi DISABLEPLATFORMCHECK=1`

But in any case, in any XTU version I can't find how to overclock the igpu.

And I suspect that the first hexadecimal pattern in the E series are different in the bios, which would be the logic that overclocks the igpu.

Here the image of XTU:

[image]

Some0nee, 2024-11-18:
Based on your output from UEFITool, it looks that the hex doesn't exist in the E580 bios. It may be under a different hex/pe32 image, or maybe isn't supported on that BIOS. Just speculation though.