digmorepaka/thinkpad-firmware-patches #56: T480 + mx150

nomorealone1, 2024-11-24:
I have T480 with i7 8650u + Nvidia mx150.
Your patch helped me a lot as I was able to undervolt CPU and make my laptop a bit quieter.
Now I wonder if it's possible to unlock advanced Nvidia menu to change current dGPU temperature limit 70°C as it starts throttling way too early.
I found this GUID: 2139B13C-E51A-4E37-B236-DD7A894B111E with reference to NVidiaOptimusAcpi in my dump, hope it can help.
[image]

nomorealone1, 2025-07-03:
Adjusted the **Temperature Limit** on my own.
vBIOS of Nvidia mx150 (as well as mx250 of ThinkPad T490) is inside the main BIOS.
Here is the instruction for UEFIPatch below:

# Nvidia vBIOS | Lenovo ThinkPad T480, maybe others | Increase mx150(GP108) temp limit from 70°C to 74°C
8B9BEF90-5FAF-46CF-A32A-7E1B7D33CC87 19 P:58C3504C4541534520504F:58C3504C45415345FFFFA1
8B9BEF90-5FAF-46CF-A32A-7E1B7D33CC87 19 P:FF000100C00B2007C00B:01000100400940094009

lprobsth, 2026-02-08:
@nomorealone1 

I tried setting the throttling temperature to 80 °C. As far as I understand the HEX values are: dec2hex(temp[°C] * 32) little endian encoded uint16 values. So for 80 °C it should be: `01000100000A000A000A`

But with this mod of the UEFI the OS (linux) can't initialize the dGPU.

Can you give me a hint? Which values did you try?

bjoern-tantau, 2026-02-10:
The CPU and GPU can easily handle temperatures above 90°. The limits are only in place to protect laps. But my desk wouldn't mind. Personally I would love to be able to push them to their limits. Let them throttle when they really reach outrageous temperatures.

nariox, 2026-02-10:
In talks with Lenovo staff, (see this [thread](https://forums.lenovo.com/t5/Other-Linux-Discussions/T480-dGPU-throttles-at-80C-also-VBIOS-version-question/m-p/5023943?page=4) - and also [this one](https://forums.lenovo.com/t5/Other-Linux-Discussions/X1C6-T480s-low-cTDP-and-trip-temperature-in-Linux/m-p/4028489?page=39#5065183) if you are so inclined), the 80C limit is imposed by the EC to comply with some UL standard (It _might_ be UL 62368-1, but I am not entirely sure) which limits the temperatures of devices in "lap mode" (as opposed to "desk mode").

The problem is compounded by the fact that the MX150 is on the same heatpipe as the CPU, so the heat of the CPU impedes/competes with some of the heat from being transferred from the GPU. I wonder if we could get better mixed-load performance (like games and such) by dynamically tuning the CPU power limits along with the GPU power limits, but this is probably a topic for a different repo.