digmorepaka/thinkpad-firmware-patches #57: WWAN patch on X395 doesn't work.

Sws142688, 2025-02-09:
Greetings everyone!!! I have an X395 laptop (2019) and after stumbling upon this repository I decided to patch the bios. But after the patch I faced the problem of 5 signals (bios protection triggered), but I accidentally bypassed it by downloading from lenovo site bios update installer .exe unzipped it, extracted the dump, brought it to 16 MB, patched and the bios protection did not work. Only then I found out that there is a script fix_vendor_hashes.py which also allows to bypass bios protection. AMD menu in bios opened, but the patch with WWAN removal does NOT work in practice on ANY bios version (tested 1.01, 1.22, 1.30). The card does not pass the test and simply cannot be initialized by the bios. Works only L850-GL (FRU: 01AX792), but I need to work card Sierra 7455EM (FRU: 01AX746) by the way it works in X380Y. I decided to self-patched the bios with my own variant of bypassing WWAN card check, extracting PE32 image from LenovoWmaPolicyDxe and through IDA disassembled and tried to understand the logic of the algorithm of WWAN card check and subsequent initialization. But I haven't managed to make the check so that it was always successful (because I don't know assembly language very well). I ask for help in this direction and other variants of ways around this problem. I will be very grateful. P. S. By the way, AMD menu functions in 1.01 and 1.30 are different. For example, in the old version there is a setting of fan control depending on the temperature.

digmorepaka, 2025-02-10:
> Sierra 7455EM

From memory, I'm pretty sure those are USB cards. 

> The card does not pass the test and simply cannot be initialized by the bios

On a thinkpad, if your device triggers a whitelist check fail, it will halt boot. If it does not trigger an unauthorized wireless card error on an unmodified UEFI then it does not use a compatible interface and the laptop cannot talk to it. 

>By the way, AMD menu functions in 1.01 and 1.30 are different

AGESA gets updated, CBS PBS menu changes, that's just how it is. 

Sorry if I missed anything, you jammed a lot of text in a very unreadable manner together, Please correct me if I understood something wrong.

Sws142688, 2025-02-11:
The most interesting thing is that the card initially did not give a list error, but could not initialize, it turns out that the interface of this card is simply not supported? On L830-EB also tested, but it is also not supported, although it gives OS boot as a card Sierra 7455EM.

So there is no way to fix it?

Sws142688, 2025-02-11:
Or what WWAN mini pci cards could be suitable for initialization? (other than the L850-GL)