digmorepaka/thinkpad-firmware-patches #59: TPM fix not working for T480 BIOS version 1.53

Liopleus, 2025-03-04:
I can patch the BIOS and apply TPM fix on my T480 with BIOS 1.49 without issue. However, when I flashed the patched BIOS with TPM fix based on 1.53, the computer produced 8 beeps on boot without display. 

Lenovo diagnostic app identified the beep as error code 0285, with the message "TCG-compliant functionality-related error (might be TPM initialization failure).

Flashing back to patched BIOS based on 1.49 fixed the issue, but it would be nice to be able to use patched BIOS based on later version.

pog5, 2025-03-06:
Same issue on T480s 20L7/20L8

joshuadyck, 2025-04-30:
Same on T480s 1.60

joshuadyck, 2025-06-25:
Look into the mentioned thread. You need to Flash to a stock bios version somewhat low, and then patch this version

Liopleus, 2025-10-29:
> [@Liopleus](https://github.com/Liopleus) does the undervolt work in 1.49?

If you only want to undervolt you can upgrade to the latest BIOS. Lenovo removed the undervolt limitation if you disable SGX. No need to mod BIOS, just use ThrottleStop or similar utilities.