digmorepaka/thinkpad-firmware-patches #63: T470 patches no boot, 0285 beep code

ArtNlk, 2025-04-25:
Laptop info:
BIOS version: N1QETA4W 1.79
BIOS date: 2024-11-05
Emb. controller version: N1QHT54W 1.36
Machine type model: 20JNS09K0Z 

I applied xx70_xx80_patches_v7.txt patches, and tried images with and without TPM fix, but in all times got same behavior - no boo with 0285 beep error code. I am still investigating this issue, but my suspicion is that i wont be able to use these patches, as when i checked with ME tool, it showed Measured boot off, but Verified boot on. Is Verified boot Boot Guard a deal breaker?
For now i will try: 

1. downgrading to 1.49 as reported in Issue #59 
2. fix_vendor_hashes.py
3. thinkpad-uefi-sign

I have little hope, as UEFITool shows that modified partitions are covered by Boot Guard - can anyone share their experience?

joshuadyck, 2025-04-30:
Having the same issue on a 20L8 T480s and BIOS 1.60...

ArtNlk, 2025-04-30:
@joshuadyck is your bootguard in verified boot too?

ArtNlk, 2025-04-30:
If you are on Linux, follow this: https://github.com/felixsinger/bootguard-status to see if you have Verified boot enabled

Not sure how to do it on Windows though - you might have to make a live linux USB drive and try it from there

joshuadyck, 2025-04-30:
I think I couldn't get iomem=relaxed to work but it tells me 

Boot guard MSR Output: 0x300000051
Measured boot: OFF
Verified boot: ON
FACB in FPFs : ON
Module revoked : OFF
NEM enabled : ON
Verified boot is enabled and ACM has enabled Cache-As-Ram.
You can't flash other firmware!

Is this bad/the issue?

joshuadyck, 2025-04-30:
Yeah, I've read the forum post now too. Seems very complicated. Do you think only for the Intel Graphics Memory upgrade option the hassle would be worth it?

ArtNlk, 2025-04-30:
I think the FB -> FF patch might be the workaround paranoidbashthot discovered - from what i understand it switches TPM into some dev/manufacturing mode which somehow leads to bootguard not being able to check bios signature. But looks like it was not in verified boot, or it was an earlier version of firmware, either bios or Intel ME (it has a separate firmware from bios itself, though on the same flash chip (at least for my case))

Downgrading bios is pretty simple, as you can get old bios downloads from lenovo website (they are in the bottom of bios upgrade utility download page)

But downgrading ME is trickier as time to time Intel would fuse a sort of minimum version version into pch, usually when vulnerabilities just like this one are found, so an attacker (even if that attacker is a legal owner of that machine!) would not be able to downgrade it to lower version firmware.

I downgraded my bios to 1.39, as was mentioned in some other issue here, but did not try to patch that, or to downgrade ME firmware due to lack of free time

Also, you can look into core/libre boot deguard - it uses another cve to patch ME firmware such that bootguard is disabled there (which allowed core/libre boot on some thinkpads with BG)

Problem is that it has configs for patching ME 11.6 but i have 11.8 - and it was patched right around that version. So i would need to understand how to properly make these configs, and pray that my version is still vulnerable

digmorepaka, 2025-05-03:
Sorry don't really have time to look into this. As for the tamper checks this is completely unrelated to bootguard, these patches occur in flash regions that are not checked by it(I think uefitool NE highlights it). Bashu's bypass turns off phoenix biosguard(this is utilized by the vendor firmware to check regions that are not checked by bootguard). On older generations the thinkpad-uefi-sign can calculate and fix the checksum for biosguard and therefore tpm also works. I believe it could be implemented for newer models but I am nowhere near competent enough with python to do so. 

https://github.com/thrimbor/thinkpad-uefi-sign/

ArtNlk, 2025-05-03:
My UEFI tool NE A70 highlights them in cyan - from what i googled that means they are protected:

[image]

Here is the bios dump from my board - GH does not allow binaries so i renamed it
[bios_t470.txt](https://github.com/user-attachments/files/20023161/bios_t470.txt)

joshuadyck, 2025-05-03:
Sorry if it's a dumb question but what is bashus bypass? Is it the Hex FB - FF one?

digmorepaka, 2025-05-03:
> My UEFI tool NE A70 highlights them in cyan - from what i googled that means they are protected:
> 
> [image]
> 
> Here is the bios dump from my board - GH does not allow binaries so i renamed it [bios_t470.txt](https://github.com/user-attachments/files/20023161/bios_t470.txt)

red / yellow is bootguard protected from what I remember (not 100% sure, been a while since I used NE)

ArtNlk, 2025-05-03:
From UEFITool source, cyan mean "Vendor fully in range"

```
switch (marking(index)) {
            case BootGuardMarking::BootGuardFullyInRange: return QBrush((Qt::GlobalColor)(markingDarkModeFlag ? Qt::darkRed    : Qt::red   )); break;
            case BootGuardMarking::VendorFullyInRange:    return QBrush((Qt::GlobalColor)(markingDarkModeFlag ? Qt::darkCyan   : Qt::cyan  )); break;
            case BootGuardMarking::PartiallyInRange:      return QBrush((Qt::GlobalColor)(markingDarkModeFlag ? Qt::darkYellow : Qt::yellow)); break;
            }
```

What can be causing the beep code error if not verified mode boot guard failing, then?

digmorepaka, 2025-05-03:
> Vendor fully in range

Exactly, that refers to it being in the vendor's tamper detection. Not Bootguard. 

https://medium.com/firmware-threat-hunting/bypass-intel-boot-guard-cc05edfca3a9

ArtNlk, 2025-05-03:
Are there any logs i can collect to analyze what goes wrong?

ArtNlk, 2025-05-04:
How do i turn off the tamper detection? Do i need to roll back the firmware?

ArtNlk, 2025-05-05:
Update: rolling back to bios version 1.48 (N1QET73W) - advanced menu patch works. I will continue testing, will report if something is unstable/broken

@digmorepaka can you suggest/recommend any tools (unpackers/disassemblers?) I can use to debug why tamper bypass does not work?

ArtNlk, 2025-05-06:
After boot it was angry about time, used this to bypass it so that os would set correct time (as date time tab is replaced): https://www.reddit.com/r/thinkpad/comments/6h1d0o/stuck_at_real_time_clock_error/

Memory settings dont work - changing memory clock ratio fails to boot and auto resets

Changing cpu power limits and ratios got me from 2.9 GHz to 3 Ghz according to /proc/cpuinfo, but seemingly only in idle (my cpu is Intel i5-6300U) - any significant workload immidiately drops it to 2.9 - are there any extra settings that unlock more/longer boost?

IGpu overclocking doesnt seem to work - only 6 extra points from glmark2

Looks like most options actually dont do anything - does anyone know actually good ones?

OMGImgreeen, 2025-10-29:
@ArtNlk how were you able to downgrade? Im currently on 1.56 and I don't think the patches work there